CNMAT/OSC #26: Max patch examples not working due to missing odot objects

sumyungguy, 2015-01-18:
I've just installed the latest odot (1.0.6rc) and OSC for Arduino (this repository). The example Max patches in the "Applications" folder won't run, due to these missing objects:

o.io.slipserial
o.io.slipserial.display.maxpat
o.io.udp

Several of the patches call for o.message (found in "deprecated"), which I guess should be replaced by o.compose and o.display?

Also o.downcast, and o.was (found in "dev") - are these ok to use?

ilzxc, 2015-01-18:
The missing objects are an unfortunate by-product of odot developments last summer. I'll take a look at these as soon as I can.

`dev` stuff is okay to use in this case. Things in `deprecated` aren't as safe, so I'll replace those with the currently supported modules.

adrianfreed, 2015-01-19:
o.io is not released currently. I will put the ones
need for the Arduino library into the repo soon.

sumyungguy, 2017-06-01:
Curious why this was closed? The Max examples still don't work.

Also a bit curious that this project isn't getting a little more love - it's a great tool for Max + Arduino, much better in my opinion than [Maxuino/Firmata](http://www.maxuino.org/). More people ought to know about it!

adrianfreed, 2017-06-01:
Thanks for the supportive words.

adrianfreed, 2017-06-01:
To specifically address this issue: the examples should be unraveled from odot dependencies unless 64-bit Windows issues with odot are going to be addressed soon.

michcan, 2021-04-12:
Hello !
I'm trying to make Max example UDPOscuino work. "o.io.udp" is the only missing object :( . "o.io.udp" is missing on the last version of odot (1.3.0-rc2). Is there a way to get the binary on windows ? Maybe i can use udp receive from max built in or other lib ?
Thank you