CNMAT/OSC #29: Unclear examples for beginners

Somnitec, 2015-05-21:
The examples are written in a way which is not easily scalable for beginners. I spend considerable time figuring out how to make this library work for me and to avoid this hassle for others, please consider adding a basic send and receive example similar to mine pasted below.
```
/*

Example send and receive OSC over serial
by Arvid Jense
21-5-2015

*/

#include <OSCBundle.h>
#include <SLIPEncodedSerial.h>
SLIPEncodedSerial SLIPSerial(Serial);

int testValue = 1337;

//basic timer
unsigned long timer = 0;             // will store last time  updated
unsigned long timerInterval = 1000;  // interval at which to time (milliseconds)

void setup() {
  SLIPSerial.begin(115200);
  pinMode(13, OUTPUT);

}

void loop() {
  //check for new messages as fast as possible
  receiveOSC();

//echo the value every timerInterval
  unsigned long currentMillis = millis();
  if (currentMillis - timer > timerInterval) {
    timer = currentMillis;
    sendOSC();
  }

}

void receiveOSC() {
  OSCMessage messageIN;
  int size;
  //receive a bundle
  if (SLIPSerial.available()) {
    while (!SLIPSerial.endofPacket())
      if ( (size = SLIPSerial.available()) > 0)
      {
        while (size--)
          messageIN.fill(SLIPSerial.read());
      }
    if (!messageIN.hasError())
    {
      messageIN.dispatch("/test/in", action);
    }
  }
}

void action(OSCMessage &msgIn) {
  int testValue = msgIn.getInt(0);
}

void sendOSC() {
  OSCMessage msg("/test/out");
  msg.add(testValue);
  SLIPSerial.beginPacket();
  msg.send(SLIPSerial); // send the bytes to the SLIP stream
  SLIPSerial.endPacket(); // mark the end of the OSC Packet
  msg.empty(); // free space occupied by message
}
```

Sensebellum, 2015-07-11:
I completely agree. I have been trying for the past 6 hours and have not been able to send a message to or from at all yet.

I am trying to do it over Ethernet - which is only more mysterious...

Might you have a simple send example for that?

TrippyLighting, 2015-07-11:
What device are you trying to send OSC messages to ?
I've been using TouchOSC to control Teensy boards  and wrote a little beginner tutorial.
Not sure if it helps.
http://trippylighting.com/teensy-arduino-ect/touchosc-and-arduino-oscuino/

Sensebellum, 2015-07-11:
I am trying to send OSC messages to my Mac BookPro. Preferably to Processing although Quartz Composer works fine.

I see a lot of examples using TouchOSC. But I am trying to strip away the layers of complexity and make it as dead simple as possible. I just cannot seem to understand what I am missing. I thought this would be a piece of cake!

Here is the thread on the Arduino Forums.

thorangutang, 2017-07-04:
+1 for some simple examples. 
I would also in the readme describe the difference between a osc message and bundle. There is as far as I can see no included code example that actually shows how to receive a message. All the arduino code examples deal with receiving bundles only. 

I wanted to control my arduino clone (adafruit feather m0 wifi) using the i-score osc sequencer that sends osc messages not bundles, and based on the examples could only make it work by using a max msp translation patch to encode the messages comming from i-score to bundles and send from max to the arduino(Wifi). 

Only much later and after much googling did I think "surely it must be possible to receive a osc message" and start emplying my limited coding experience to decode the API. 

It wass of course possible, Please make the difference between a bundle and message clear for those checking out the library for the first time!!

seanr2109, 2018-01-14:
How easy is it to code something that uses say 6/8 inputs from buttons, on a Leonardo Ethernet board.
I want to build a device that has some buttons that each send a different OSC command out from each button on ethernet.

FigrHed, 2018-05-01:
This example does not work for me, has anyone else tried it?
Thank you for the simple example that uses 'Serial' in SLIPSerial instead of 'Serial1' as I have an uno not a mega or leonardo. 
I am fairly confident that I am sending OSC to my arduino (though I haven't worked out how to test it yet) but I don't think my arduino is sending back yet and just don't know how to troubleshoot this sort of problem. Has anyone else had any progress?