CNMAT/OSC #37: release current master for automatic availability in Arduino Library Manager

marcobrianza, 2015-11-17:
The Arduino team has linked the library for automatic inclusion in the IDE Library manager. 
To get things working a new release is needed. Simple guidelines are available here:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

When fixed, within one hour from the release the library manger will show the lib in the IDE.

marcobrianza, 2015-11-17:
this is the comment from Arduino 
https://github.com/arduino/Arduino/issues/4169

bltzr, 2018-01-14:
I'm seconding this - would you agree to release the library through Arduino's library manager, @adrianfreed ? That would be very useful!
AFAIK, you just need to add a new tag to the repo at the current state.
Thanks in advance!

bltzr, 2018-01-22:
Hi @adrianfreed !

Is there any reason why you're holding this release ? Like things you'd like to check first or something ?
Or is it because it's not working on all Arduinos ? (@avilleret reported to me last week that he had some problems with an Arduino uno), and thus it's just released via teensyduino ?
Just trying to figure out... sorry for insisting on this.

With all due respect, I think that's really a pity that the OSC lib isn't available in the standard Arduino lib manager if that's just a question of creating a release (which requires no more than a 30 seconds action on https://github.com/CNMAT/OSC/tags ) - it really doesn't help to make this available (e.g. when recommending it to students, in my case...)

Thanks for all the nice work !