CNMAT/OSC #39: OSCData doesn't compile on Galileo

tambien, 2015-11-25:
The fix is seems pretty easy, we just need to find the right `ifdef` for the Galileo (and probably also the Intel Edison) for [this](https://github.com/CNMAT/OSC/blob/master/OSCData.h#L96) line (and the definition in the cpp file). This is probably because the `ints` are `int_32` on these platforms, so it grumbles that there is a redefinition. 

Maybe, is there a case for just getting rid of the `int` OSCData type constructor and just keeping `int_16` and `int_32`? Moving forward there are going to be more 32bit platforms so that ifndef is probably going to keep growing. But i'm not sure right now if we would break anything if we removed the normal `int` constructor.

adrianfreed, 2015-12-10:
can we do some clever ifdefs with #if sizeof(int)!=sizeof(int_32)

tambien, 2015-12-10:
I don't think we can use `sizeof` in the preprocessor. ([stack overflow](http://stackoverflow.com/questions/4079243/how-can-i-use-sizeof-in-a-preprocessor-macro))

adrianfreed, 2018-03-05:
STM32 is producing this problem too.