CNMAT/OSC #40: can't decode blob since 698eb704bbd3c419ebf55441c4bae5a3f65332c2

avilleret, 2016-01-05:
since 698eb704bbd3c419ebf55441c4bae5a3f65332c2, `getBlob` method always returns 0
and the buffer is not filled with values.
before that commit the first 4 bytes were filled with the blob size (which is a bug that should be fixed).
now is doesn't work at all.

here is the code I'm using for testing : https://gist.github.com/avilleret/4b020ea34bd2f5bc1692
Tested on Arduino UNO and Teensy 3.2

avilleret, 2016-01-06:
reverting 698eb704bbd3c419ebf55441c4bae5a3f65332c2 from 376b8e7389ca8819d18c0690a07680dfb615e123 doesn't work

adrianfreed, 2016-01-06:
thanks. first OSC item on my 2016 todo list.

dimitre, 2016-08-29:
Here I can decode the first blob, but not the second position.
msg.getBlob always returns zero bytes even when msg.getDataLength(1) returns a valid message length.
Thanks

dimitre, 2016-08-29:
Looking more closely today I've noticed Teensy receive the first blob argument OK if it has a size under 1422 bytes. It stops receiving once the message is bigger.
Thanks

adrianfreed, 2016-08-29:
It may have run out of memory.
Which Teensy? Teensy 3 or 2?

dimitre, 2016-08-29:
@adrianfreed Teensy 3.2
I supposed the same, will try to make some function showing the free memory to check that.

dimitre, 2016-08-29:
@adrianfreed I've checked and it has plenty of free memory while parsing that
```
len:1417
freeMemory()=47548
```

dimitre, 2016-08-29:
I think https://github.com/ntruchsess/arduino_uip library is to blame. Dont have another ethernet adaptor to test it further. Thanks

adrianfreed, 2016-08-29:
Oh yes, the number you mentioned may be close to the  MTU for the library/hardware you are using. https://en.wikipedia.org/wiki/Maximum_transmission_unit

avilleret, 2017-08-18:
> first OSC item on my 2016 todo list.

We are now far in 2017, how your 2016 todo list is going ? :-)

bltzr, 2018-01-11:
I guess this is fixed now