CNMAT/OSC #44: Time Tag synchronization, does it already work?

kasperkamperman, 2016-03-14:
I'm researching a way to synchronise different Arduino's (well actually Particle Photons) on a network. In this case to display animations at the same time. Research NTP I came across the OSC time tag. 

However I can only find literature from about 10 years ago. Are there working implementations (preferably Arduino)?

adrianfreed, 2016-03-14:
There is a simple technique we wrote a paper about that may help.
Unfortunately to my knowledge it hasn't been done on Particle/Arduino yet.

http://quod.lib.umich.edu/cgi/p/pod/dod-idx/simple-synchronisation-for-open-sound-control.pdf?c=icmc;idno=bbp2372.2015.045

kasperkamperman, 2016-03-15:
Hi Ardian. 

A very interesting read. Since you already tested this with WiFi nodes, is there already code available that might be portable to Arduino/Particle?

You mention that all the boards have a physical clock. Do you mean that it's an RTC clock module? Or could it be the millis/micros counter in an Arduino?