CNMAT/OSC #46: Malformed packets when used together with WiFi101

Axagthoth, 2016-04-08:
Hi,

I'm trying to use this library with the new Arduino MKR1000 board, which uses the WiFi101 library (a modified version of the WiFi library) to communicate.
Receiving messages on the board works like a charm, I've tried it both with TouchOSC and Processing, and have managed to control one of the PWM pins on the board remotely.
The problem appears when trying to send messages from the board to the computer: no matter what I try, I can't seem to create "correct" OSC messages that my receivers can interpret.
Checking with the example codes I think the issue can only lie with one of the three following calls:

Udp.beginPacket(remIP, remPort);
msg.send(Udp);
Udp.endPacket();

I've reached this conclusion because comparing the code of WifiUDP with that of EthernetUDP (with which the same code works) I've seen that the code is quite different, but I'm not savvy enough to identify which pieces of the code might be critical.
The question is: **do any of these functions add/require something to/from the UDP packet**? The reason I'm asking this is that going deeper and analyzing the packets with Wireshark, I've realized that (at least) precisely 4 bytes are wrong, so maybe something is screwing the 4-padding requirement of OSC up? Here are the images of the capture that make me suspect this (I'm trying to send "layer1/clip1/connect 1"):

[image]

[image]

[image]

Any help and ideas would be greatly appreciated. Thanks a lot in advance!

adrianfreed, 2016-04-08:
thanks for the update on the new Arduino board. We are waiting for ours to arrive and will investigate as soon as possible.

That capture data is indeed odd. I am looking at related wifiudp.h code and notice that the predefined packet size is way too small for many applications (24 bytes).
I will review this with the new wifi udp Arduino code. One strategy (which we should have used in the
examples) is to test (or report) the return value every call in that sequence. It would be good
to know when the packet one was building up is too large.

Axagthoth, 2016-04-09:
Hi Adrian! Thanks for answering so quickly. And thanks for maintaining this wonderful library and your work at the CNMAT! I'm pretty sure you'll love the MKR1000, it's an amazing little board. I imagine you are interested in getting it to work with OSC over WiFi too.
I posted the same issue on the WiFi101 side of things, [here](https://github.com/arduino-libraries/WiFi101/issues/56#issuecomment-207531235). I'm not sure, but I think you can close the issue on the Oscuino side, since I believe the problem lies with the WifiUDP code, like you already said. You'll see I tried sending the same message raw over UDP, like this (no oscuino involved):

```
/layer1/clip1/connect 1

char rawOSC[] = { 47, 108,  97, 121,
                 101, 114,  49,  47,
                  99, 108, 105, 112,
                  49,  47,  99, 111,
                 110, 110, 101,  99,
                 116,   0,   0,   0,
                  44, 105,   0,   0,
                   0,   0,   0,   1
                   };
  Udp.beginPacket("192.168.1.6", remPort);
  Udp.write(rawOSC);
  Udp.endPacket();
```

And curiously enough, I got the same strange packet traces I posted above (an LLC packet with exactly 17 bytes of data). This has moved my suspicions towards the write() or beginPacket() functions. I'll also look into what you said about the predefined size being 24. It really isn't much, when even simple OSC messages like the one above occupy 32 bytes.

Axagthoth, 2016-04-19:
The WiFi101 library has been updated with a transmit buffer, and now everything works fine!

adrianfreed, 2016-05-01:
Yeah!