CNMAT/OSC #52: Getting an error when I use OSCMessage.add() to add a string

colonialT, 2016-08-16:
When I try to use a variable with a string value I get an error. 
Error code: "note:   no known conversion for argument 1 from 'String' to 'const OSCData&'"

Not sure if its an issue with the ESP8266 board using the Arduino IDE, the OSC library or my own inexperience.  Any help, greatly appreciated. Thank you

Using Arduino IDE 1.6.9 
OSC Library v1.3.3
ESP8266 12e (Esp8266 Board plug in for Arduino IDE v 2.3.0)

Example code from project below.

Function that is giving me the error.  The function just takes the variable "doorFrame" which is an integer, converts it to a string and sends it to a label on TouchOSC. 

`void sendDoor(OSCMessage &msg, int addrOffset) {
  OSCMessage doorMsg("/Door/getSize");
  String _doorFrame = String(doorFrame); //Currently getting an error when I feed a string to .add().
  doorMsg.add(_doorFrame);
  
  OSCMsgSend(doorMsg, touchOscIp);
}`

I have also tried just making a new string variable and passing that but it also gives me an error.

`void sendDoor(OSCMessage &msg, int addrOffset) {
  OSCMessage doorMsg("/Door/getSize");
  String myString = "I'm a string"; //Currently getting an error when I feed a string to .add().
  doorMsg.add(myString);
  
  OSCMsgSend(doorMsg, touchOscIp);
}`

TrippyLighting, 2016-08-16:
You should be able to write the integer directly to the label without converting it into a string.

colonialT, 2016-08-18:
TrippyLighting, that did indeed work.  For this project an integer will do but I still cant get a string to work.  I'm rather new to GitHub, would you recommend I post this to the ESP8266 github page? 

Also, your tutorial using an Arduino, ethernet shield and the OSC library to communicate with TouchOSC has been a fantastic guide in getting me set up with my project.  Many thanks for the work you put in to that.

Cheers