CNMAT/OSC #53: Dispatch doesn't seem to be working on my Wemos D1R2

gavspav, 2016-10-31:
I'm receiving bundles ok but I never get to the function specified in the arguments of dispatch.
I am running the ESP8266Receive Bundle sketch. Here are the only bits I've altered.
Serial monitor reports 'dispatched' but I never get a 'received' message.

```
void led(OSCMessage &msg) {
  Serial.println("received");
  ledState = msg.getInt(0);
  Serial.println("received");
  digitalWrite(BUILTIN_LED, ledState);
  Serial.print("/led: ");
  Serial.println(ledState);
}

void loop() {
  OSCBundle bundle;
  int size = Udp.parsePacket();

  if (size > 0) {

    while (size--) {
      bundle.fill(Udp.read());
    }
    if (!bundle.hasError()) {
      
      bundle.dispatch("/led", led);
      Serial.println("dispatched");
    } else {
      error = bundle.getError();
      Serial.print("error: ");
      Serial.println(error);
    }
  }
}
```

lhanneus, 2016-12-17:
Same problem for me.

Leamucho, 2017-02-14:
Yes, the same problem to me also, only on SlipSerial... the function never runs....
the library is quite abandoned... have you seen or know about what happened to the CNMAT page or maybe the institution itself?

christophejourdan, 2018-01-07:
Still not working for me too

jonasfehr, 2018-06-05:
have you found a fix for the issue?

VanDerSloth, 2022-04-02:
Has anyone ever solved this? I have the same problem with a D1 mini. "match" doesn't seem to be working either. I do receive a message, I can even read out the value, I'm just unable to use the address to my benefit.

stevefal, 2022-04-13:
I'm trying to remember if I needed to do something else to get this working, but it does for me. This code snippet looks like the sample, and it works:

```
void receiveOSC(){
  OSCBundle bundle;
  int size = Udp.parsePacket();
  
  if (size > 0) {
    while (size--) {
      bundle.fill(Udp.read());
    }
    if (!bundle.hasError()) {
      bundle.dispatch("/dev/0/0/mon", cmdVol);
      bundle.dispatch("/out/fvOutCS+0/out/ol", cmdVU1);
      bundle.dispatch("/out/fvOutCS+0.1/out/ol", cmdVU2);
      bundle.dispatch("/out/fvOutCS+1/out/ol", cmdVU3);
      bundle.dispatch("/out/fvOutCS+1.1/out/ol", cmdVU4);
      bundle.dispatch("/out/fvOutCS+2/out/ol", cmdVU5);
      bundle.dispatch("/out/fvOutCS+2.1/out/ol", cmdVU6);       
    } else {
      myerror = bundle.getError();
    }
  }
}

void cmdVol(OSCMessage &msg) {
  setDisplaySleep(0);                 //unsleep
  currentVolume = msg.getFloat(0);
}

void cmdVU1(OSCMessage &msg){
  messageReceived();
  vu[0] = msg.getFloat(0);
}

  //etc...
```

stevefal, 2022-04-13:
Not sure if this check-in has any bearing: https://github.com/CNMAT/OSC/commit/c9f33d2bf45e3b7edbd6b26c946d1eff28d9689e