CNMAT/OSC #57: ESP8266 Example is not working?

JonasVorwerk, 2017-04-12:
Experimenting with an ESP8266, sending and receiving OSC messages with the included ESP examples:

https://github.com/CNMAT/OSC/blob/master/examples/ESP8266ReceiveBundle/ESP8266ReceiveBundle.ino and https://github.com/CNMAT/OSC/blob/master/examples/ESP8266sendMessage/ESP8266sendMessage.ino

Sending works straight away but receiving data doesn't seem to work. Does anybody know how to get this working?

I'm sending and receiving OSC data from Max7 with the default "udpsend"  object and I'm pretty sure it works.

miyamonz, 2017-05-29:
me too.
sending works but receiving doesn't work.

```arduino
void loop() {
  OSCBundle bundle;
  int size = Udp.parsePacket();

  if (size > 0) {
    while (size--) {
      bundle.fill(Udp.read());      
    }
          Serial.println(bundle.size()); // => 0

    if (!bundle.hasError()) {
      bundle.dispatch("/led", led);
    } else {
      error = bundle.getError();
      Serial.print("error: ");
      Serial.println(error);
    }
  }
}```

JonasVorwerk, 2017-05-29:
If anybody has a working version or ideas to get it working, I’m still very interested and any help would really be appreciated!

JonasVorwerk, 2017-05-31:
Why the example works with bundles is a mystery, this seems so work. Enjoy

void led(OSCMessage &msg) {
  ledState = msg.getInt(0);
  digitalWrite(BUILTIN_LED, ledState);
  Serial.print("/led: ");
  Serial.println(ledState);
}

void loop() {
  OSCMessage msg;
  int size = Udp.parsePacket();
  if (size > 0) {
    while (size--) {
      msg.fill(Udp.read());
    }
    if (!msg.hasError()) {
      msg.dispatch("/led", led, 0);
      msg.empty();
    } else {
      error = msg.getError();
      Serial.print("error: ");
      Serial.println(error);
    }
  }
}

miyamonz, 2017-06-01:
It works! thank you.

closed 2018-03-03 by adrianfreed