CNMAT/OSC #66: OSC Objects Copy/Destruction - can cause crash

GIPdA, 2017-09-20:
Hi,

Constructors and destructors are not done correctly in (almost?) all OSC objects (no copy constructors), so if you try passing around objects by copy (no pointer or reference), you end up freeing already-freed memory -> crash.

Adding some sort of implicit sharing would be the best option to fix the issue. The workaround is of course to use references or pointers and be careful of not doing any copy.

I'll try to push a fix to that asap.

adrianfreed, 2018-01-12:
Interesting challenge. On early Arduinos there is so little memory people tend not to copy objects.

GIPdA, 2018-01-12:
Indeed. But for non-experienced developers, it's easy to do copies by mistake like passing objects to functions by copy instead of by reference to process a message.
Everything is fine until the first object is deleted. And that could be quite difficult to debug.

A simple fix or workaround I guess would be to delete or make private the copy constructors (and a note in the readme). Or use that old school reference counter.

closed 2018-03-03 by adrianfreed