CNMAT/OSC #68: Issues while compiling current build

freadZdead, 2017-11-16:
Hi there,

Running Arduino 1.8.5, for an Arduino UNO, trying to use/include these libraries in a project, but currently failing at compile stage of even the integrated example OSCBundle Test.

Any idea what could be going wrong/is it easily fixable? error messages below:
```

In file included from /Users/mc/Documents/Arduino/libraries/OSC/OSCBundle.h:29:0,
                 from /Users/mc/Documents/Arduino/libraries/OSC/test/OSCBundle_test/OSCBundle_test.ino:2:

/Users/mc/Documents/Arduino/libraries/OSC/OSCMessage.h: In instantiation of 'OSCMessage& OSCMessage::add(T) [with T = int]':

/Users/mc/Documents/Arduino/libraries/OSC/test/OSCBundle_test/OSCBundle_test.ino:9:31:   required from here

/Users/mc/Documents/Arduino/libraries/OSC/OSCMessage.h:136:34: error: call of overloaded 'OSCData(int&)' is ambiguous

   OSCData * d = new OSCData(datum);
                                  ^

/Users/mc/Documents/Arduino/libraries/OSC/OSCMessage.h:136:34: note: candidates are:

In file included from /Users/mc/Documents/Arduino/libraries/OSC/OSCMessage.h:29:0,

                 from /Users/mc/Documents/Arduino/libraries/OSC/OSCBundle.h:29,

                 from /Users/mc/Documents/Arduino/libraries/OSC/test/OSCBundle_test/OSCBundle_test.ino:2:

/Users/mc/Documents/Arduino/libraries/OSC/OSCData.h:104:5: note: OSCData::OSCData(boolean)

     OSCData  (boolean);
     ^

/Users/mc/Documents/Arduino/libraries/OSC/OSCData.h:103:2: note: OSCData::OSCData(OSCData*) <near match>

  OSCData (OSCData *);
  ^
/Users/mc/Documents/Arduino/libraries/OSC/OSCData.h:103:2: note:   no known conversion for argument 1 from 'int' to 'OSCData*'

/Users/mc/Documents/Arduino/libraries/OSC/OSCData.h:100:2: note: OSCData::OSCData(double)

  OSCData (double);
  ^

/Users/mc/Documents/Arduino/libraries/OSC/OSCData.h:99:2: note: OSCData::OSCData(float)

  OSCData (float);
  ^

/Users/mc/Documents/Arduino/libraries/OSC/OSCData.h:98:5: note: OSCData::OSCData(unsigned int)

     OSCData (unsigned int);
     ^

/Users/mc/Documents/Arduino/libraries/OSC/OSCData.h:94:2: note: OSCData::OSCData(int32_t)

  OSCData (int32_t);
  ^
/Users/mc/Documents/Arduino/libraries/OSC/OSCData.h:90:2: note: OSCData::OSCData(const char*) <near match>

  OSCData(const char * s);
  ^

/Users/mc/Documents/Arduino/libraries/OSC/OSCData.h:90:2: note:   no known conversion for argument 1 from 'int' to 'const char*'
/Users/mc/Documents/Arduino/libraries/OSC/OSCData.h:65:5: note: OSCData::OSCData(char)

     OSCData(char t);
     ^

/Users/mc/Documents/Arduino/libraries/OSC/OSCData.h:55:7: note: constexpr OSCData::OSCData(const OSCData&)

 class OSCData 
       ^

exit status 1

Error compiling for board Arduino/Genuino Uno.

```

Thanks for looking into this!

Cheers,

freadZdead

adrianfreed, 2018-03-02:
Yeah, something broke in one of the pull requests .

adrianfreed, 2018-03-03:
I just fixed this.

closed 2018-03-03 by adrianfreed