CNMAT/OSC #77: examples using i2c sensors?

dappertutto, 2018-02-21:
I've been working with the esp8266 examples and now trying to send bundles of i2c sensor data going over wifi. Are there any sketches that demonstrate this anywhere?

adrianfreed, 2018-02-22:
it took me 15seconds with a google search to find this: http://apps.incalcando.com/2017/05/22/sensor-data-transmitted-as-osc-over-wifi-huzzah-esp8266/

dappertutto, 2018-02-22:
Thanks. I was at the site yesterday and as far as i could tell, it had little to do w/my sketch. I've learned some things in the meantime thanks to some helpful support and I have a sketch that works. Is there a place on the github where i could link to some sketches, or is this not really what goes on here?

adrianfreed, 2018-02-23:
I am glad you have something that works.

Getting something that works into something of value as a library example is a bit of a chore, because to be useful to people it has to be tested with the ever
growing set of different Arduino related boards. As a result the OSC examples are rather few and tend to use built-in features instead of boards people have added
(via i2c etc.). If you used a really common i2c device which has a solid Arduino support library (e.g. Adafruit stuff), it may be valuable to add. The guy that
does this really well is Paul at PJRC of Teensy fame. His libraries tend to be the best because he tests on a lot of hardware, desiging special PCBs that
exercise each interesting library peripheral chip. If the i2c device you are using has a solid library from him, it may be worth including your example.

There may be a case also here to put your work into a different repo than the CNMAT/OSC one. One way to think about it is. “Do I want to respond to e-mail
from strangers using the OSC Arduino library who are trying to get something from my example” or “Do I want to respond to e-mail from strangers doing something like
what I was doing?”.  Consider the case of someone (like me soon) who has an ESP32 board and is trying to get your code to work…