CNMAT/OSC #79: infinite loop while reading

sergiobd, 2018-03-15:
Hi, 

I am trying to control a LED strip using FastLED and the Arduino OSC library. Apparently, the code gets stuck in the reading loop and therefore the color on the strip never updates. If I send a message (which I am doing using PureData), the code gets unstuck for a frame and then it gets stuck again. 

This is my code:

```
#include <OSCBundle.h>
#include <OSCBoards.h>

#ifdef BOARD_HAS_USB_SERIAL
#include <SLIPEncodedUSBSerial.h>
SLIPEncodedUSBSerial SLIPSerial( thisBoardsSerialUSB );
#else
#include <SLIPEncodedSerial.h>
 SLIPEncodedSerial SLIPSerial(Serial);
#endif

#include "FastLED.h"

FASTLED_USING_NAMESPACE

int led = LED_BUILTIN;

// * * * * * LED * * * *

#define DATA_PIN    11
//#define CLK_PIN   4
#define LED_TYPE    WS2812B
#define COLOR_ORDER GRB
#define NUM_LEDS    30

CRGB leds[NUM_LEDS];

#define BRIGHTNESS          96
#define FRAMES_PER_SECOND  120 

bool ledOn;

void setup() {

  // * * * * LED * * * * 
  
  delay(3000); // 3 second delay for recovery
  
  // tell FastLED about the LED strip configuration
  FastLED.addLeds<LED_TYPE,DATA_PIN,COLOR_ORDER>(leds, NUM_LEDS).setCorrection(TypicalLEDStrip);

  // set master brightness control
  FastLED.setBrightness(BRIGHTNESS);

  // * * * * SERIAL * * * * 
  // Subir velocidad
  Serial.begin(9600);
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(led, HIGH);

  SLIPSerial.flush();
}

// List of patterns to cycle through.  Each is defined as a separate function below.
typedef void (*SimplePatternList[])();
SimplePatternList gPatterns = { rainbow, rainbowWithGlitter, confetti, sinelon, juggle, bpm };

uint8_t gCurrentPatternNumber = 0; // Index number of which pattern is current
uint8_t gHue = 0; // rotating "base color" used by many of the patterns
  

void loop() {

  // * * * * * LED * * * * 
  // Call the current pattern function once, updating the 'leds' array
  gPatterns[gCurrentPatternNumber]();

  // send the 'leds' array out to the actual LED strip
  FastLED.show();  
  // insert a delay to keep the framerate modest
  FastLED.delay(1000/FRAMES_PER_SECOND); 

  // do some periodic updates
  EVERY_N_MILLISECONDS( 20 ) { gHue++; } // slowly cycle the "base color" through the rainbow
  
  //EVERY_N_SECONDS( 10 ) { nextPattern(); } // change patterns periodically

  
  // * * * * *  OSC RECEIVE * * * * *
  OSCMessage msg;

  while(!SLIPSerial.endofPacket()){
    int size = SLIPSerial.available();
    digitalWrite(led, LOW);
    if (size > 0){
      //fill the msg with all of the available bytes
      while(size--){
        msg.fill(SLIPSerial.read());
      }
    }
  }
  digitalWrite(led, HIGH);

  // * * * * *  OSC ROUTE * * * * *
  msg.dispatch("/led", ledFunction);

}

void ledFunction(OSCMessage &msg){
  //digitalWrite(led, LOW);
  nextPattern();
  }

// * * * * * LED COLOR FUNCTIONS * * * *   

#define ARRAY_SIZE(A) (sizeof(A) / sizeof((A)[0]))

void nextPattern()
{
  // add one to the current pattern number, and wrap around at the end
  gCurrentPatternNumber = (gCurrentPatternNumber + 1) % ARRAY_SIZE( gPatterns);
}

void rainbow() 
{
  // FastLED's built-in rainbow generator
  fill_rainbow( leds, NUM_LEDS, gHue, 7);
}

void rainbowWithGlitter() 
{
  // built-in FastLED rainbow, plus some random sparkly glitter
  rainbow();
  addGlitter(80);
}

void addGlitter( fract8 chanceOfGlitter) 
{
  if( random8() < chanceOfGlitter) {
    leds[ random16(NUM_LEDS) ] += CRGB::White;
  }
}

void confetti() 
{
  // random colored speckles that blink in and fade smoothly
  fadeToBlackBy( leds, NUM_LEDS, 10);
  int pos = random16(NUM_LEDS);
  leds[pos] += CHSV( gHue + random8(64), 200, 255);
}

void sinelon()
{
  // a colored dot sweeping back and forth, with fading trails
  fadeToBlackBy( leds, NUM_LEDS, 20);
  int pos = beatsin16( 13, 0, NUM_LEDS-1 );
  leds[pos] += CHSV( gHue, 255, 192);
}

void bpm()
{
  // colored stripes pulsing at a defined Beats-Per-Minute (BPM)
  uint8_t BeatsPerMinute = 62;
  CRGBPalette16 palette = PartyColors_p;
  uint8_t beat = beatsin8( BeatsPerMinute, 64, 255);
  for( int i = 0; i < NUM_LEDS; i++) { //9948
    leds[i] = ColorFromPalette(palette, gHue+(i*2), beat-gHue+(i*10));
  }
}

void juggle() {
  // eight colored dots, weaving in and out of sync with each other
  fadeToBlackBy( leds, NUM_LEDS, 20);
  byte dothue = 0;
  for( int i = 0; i < 8; i++) {
    leds[beatsin16( i+7, 0, NUM_LEDS-1 )] |= CHSV(dothue, 200, 255);
    dothue += 32;
  }
}
```

sergiobd, 2018-03-15:
Elaborating a little bit more about this, the variable `rstate` (in SlipEncodedSerial.cpp, lines 20 and following) starts as CHAR, therefore, the function returns `false`. This means the while loop 

` while(!SLIPSerial.endofPacket()){...}`

will run forever.

sergiobd, 2018-03-15:
I changed the reading code to:

```
  int size = SLIPSerial.available();
  while(!SLIPSerial.endofPacket() && size > 0 ){ 
    int size = SLIPSerial.available();
      //fill the msg with all of the available bytes
      while(size--){
        msg.fill(SLIPSerial.read());
      }
  }
```
and this apparently did the trick. I will give it a try. If this is the best workout to the problem I would suggest an update of the docs.

adrianfreed, 2018-03-15:
>   while(!SLIPSerial.endofPacket() && size > 0 ){ // Sale cuando endOfPacket es TRUE

Yes, the code you built from is synchronous, it busy waits until a packet arrives.
Your new code gives control back to the led updating when no new serial data is available.
 Presumably you check for the full packet to arrive before parsing it.

A better way to document this use case would probably be to include more
examples. Examples that don’t require particular additional hardware are better. Perhaps
something that changes the blink rate of the built-in led will make the point well.

manchoz, 2019-12-16:
@sergiobd, @adrianfreed please, see PR https://github.com/CNMAT/OSC/pull/107.

PR #107 (2020-01-20): Add examples for running interleaved tasks in receiving loop.
Closes #79.

I just added a couple of examples on how to refactor the receiving loop in order to be able to process/execute other tasks while waiting for incoming OSC data.

Sender controls the on/off rate at which the Receiver will blink an LED while waiting for more data incoming.
closed 2020-01-20 by adrianfreed