CNMAT/OSC #80: Error triggered when receiving bundles but message ok

natcl, 2018-04-20:
When receiving bundles over serial (using slip serial) I often get errors (INVALID_OSC)
I traced it to this function:
https://github.com/CNMAT/OSC/blob/master/OSCBundle.cpp#L280
The strange thing is that I can still decode the data fine.
Any ideas ?

Here is the code:

```

#include <OSCBundle.h>
#include <OSCBoards.h>

#ifdef BOARD_HAS_USB_SERIAL
#include <SLIPEncodedUSBSerial.h>
SLIPEncodedUSBSerial SLIPSerial( thisBoardsSerialUSB );
#else
#include <SLIPEncodedSerial.h>
 SLIPEncodedSerial SLIPSerial(Serial1);
#endif

void setup() {
  SLIPSerial.begin(115200);
  Serial.begin(115200);
}

uint8_t processOSC (void)
{
  OSCBundle  bundleIN;
  OSCMessage messageIN;
  int size;
  OSCErrorCode error;
  //If there is serial data available, read it out
  if (SLIPSerial.available())
  {
    // static unsigned long microsTime = millis();
    while (!SLIPSerial.endofPacket())
    {
      if ((size = SLIPSerial.available()) > 0)
      {
        while (size--)
        {
          bundleIN.fill(SLIPSerial.read());
        }
      }
    }
    
    if (bundleIN.hasError()) {
      Serial.println(bundleIN.getError());
    }
    
    bundleIN.route("/TEST", TEST);
    
    bundleIN.empty();
  }
}

void TEST(OSCMessage & msg, int addrOffset)
{
  Serial.println("TEST");
}

void loop() {
  processOSC();
}
```

adrianfreed, 2018-04-20:
It would be easier to help you if we had a dump of the OSC message/bundle being processed.

natcl, 2018-04-20:
On the sending side or the receiving side ?

natcl, 2018-04-20:
I'm not sure if this is the good way to debug but I'm printing the contents of the bundle byte by byte:
```
byte a = SLIPSerial.read();
Serial.print(a);
bundleIN.fill(a);
```
here's what I get:

```
47
84
69
83
84
0
0
0
44
102
0
0
63
128
0
0
Error

47
84
69
83
84
0
0
0
44
102
0
0
63
128
0
0
Message sent
```

Seems like the bundle is identical both when it errors and when it passes.

natcl, 2018-04-20:
After further analysis I found the issue:

I'm not sending a bundle, I'm sending a message which doesn't have a timetag so `incomingMessageSize` never gets set.  Since it never gets set is is not initialized with a [https://github.com/CNMAT/OSC/blob/8b60603b23c9014ad8c2b2afd02e5bba1188d253/OSCBundle.h#L70](value) and becomes a random value.
I understand I should be sending a bundle if I'm expecting a bundle on the Arduino side but perhaps incomingMessageSize should be initalized so that it always fails when a message is sent ?

adrianfreed, 2018-04-20:
> I'm not sending a bundle, I'm sending a message which doesn't have a timetag so incomingMessageSize never gets set. Since it never gets set is is not initialized with a https://github.com/CNMAT/OSC/blob/8b60603b23c9014ad8c2b2afd02e5bba1188d253/OSCBundle.h#L70 and becomes a random value.

thanks. Yes, I had just figured out that this was the bundle handling of a message (not a bundle).
I will carefully review the code for this case.