CNMAT/OSC #106: OSC and Arduino Uno WiFi Rev2

educhemin, 2019-12-09:
Hello,

It seems OSC library 1.3.5 doesn't work with Arduino Uno WiFi Rev2.

Inserting `#include <OSCMessage.h>` in my sketch gives me the message :

> Arduino: 1.8.10 (Mac OS X), Board: "Arduino Uno WiFi Rev2, ATMEGA328"
> 
> Multiple libraries were found for "SPI.h"
>  Used: /Users/emmanuel/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/libraries/SPI
> Multiple libraries were found for "WiFiNINA.h"
>  Used: /Users/emmanuel/Documents/Arduino/libraries/WiFiNINA
> Multiple libraries were found for "OSCMessage.h"
>  Used: /Users/emmanuel/Documents/Arduino/libraries/OSC
> In file included from /Users/emmanuel/Documents/Arduino/libraries/OSC/SLIPEncodedSerial.cpp:1:0:
> /Users/emmanuel/Documents/Arduino/libraries/OSC/SLIPEncodedSerial.h:11:10: fatal error: HardwareSerial.h: No such file or directory
>  #include <HardwareSerial.h>
>           ^~~~~~~~~~~~~~~~~~
> compilation terminated.
> exit status 1
> Error compiling for board Arduino Uno WiFi Rev2.

It doesn't seem to be a SPI or WifiNINA issue, they work fine without OSC...

Any solution ? Any plan to update ?
Thanks !

sstaub, 2020-01-19:
see PR #104

closed 2020-01-20 by adrianfreed

educhemin, 2020-01-25:
Thanks a lot !

Emmanuel Duchemin
-----------------------------
+33 (0)6 10 25 27 92
e.duchemin@online.fr