CNMAT/OSC #111: ESP8266 Module Won't Connect to Network

CLK88, 2020-03-01:
I followed the hookup guide here for the module: https://www.instructables.com/id/Add-WiFi-to-Arduino-UNO/ 

I changed ESP8266sendMessage example sketch with my network name, password, ip, and ports. When I upload it, in the serial monitor all I see is: 

Connecting to (networkName) 
............................. (ect) 

Not sure what else to do to get it to work.

adrianfreed, 2020-03-02:
I would need more context to help. A lot of people get stuck and eventually find that their firewall is in the way…

CLK88, 2020-03-02:
Firewall is disabled, and I double checked to be sure. 
What more do you need?

adrianfreed, 2020-03-02:
Have you run other communication sketches to test your setup which don’t involve the OSC library, e.g. a basic ping test?

CLK88, 2020-03-06:
I started having other issues and ended up completely uninstalling and re-installing the Arduino IDE. At some point during the process of trying to fix other problems, this example now works with my Feather. But for my project I want to use the generic module connected to an arduino and I'm having problems uploading any code to it now. But that's a separate issue.