CNMAT/OSC #114: Support for Teensy 4.x

sstaub, 2020-05-24:
A support for Teensy 4.0 and 4.1 is missing.

adrianfreed, 2020-05-25:
I will take a look at this when a 4.1 arrives next week.

sstaub, 2020-05-25:
As an advice, Teensy have new macro definitions for each board, e.g.
ARDUINO_TEENSY41, ARDUINO_TEENSY40 ... also for the older boards.
The microcontroller is the same for 4.0 and 4.1

MaurinElectroTextile, 2021-01-09:
Happy new year Adrian!
If you make any progress for the Teensy 4.* support it will be useful for me to ;-)

Druuka, 2022-01-11:
I'm also having issues with teensy 4.1 and the library.  Would love to have it fully supported.  I'm currently having issues routing messages after they have been parsed out of slip encoding.  I have the same sketch working on an arduino ethernet without issue.

sstaub, 2022-01-11:
@Druuka In the moment nothing happens, therefore I write my own implementation for the next version of my eOS library for ETC consoles but it will take some time if it is ready for publishing.

fcana, 2023-12-08:
hi @sstaub i had a look on your eOS project and it's impressive. i'm trying to build something similar but with midi support (using usb host) so teensy 4.1 is now the only shot i have... did you find a way to work with sliposc on teensy 4.1? 
@wagne342 is that issue still open since so long time? tnx

arnoson, 2023-12-08:
@fcana I'm using this library with a Teensy 4.1 and a slightly modified Slip Serial: https://github.com/miwos/miwos/blob/main/bridge/device/lib/Bridge/include/SlipSerial.h
Usage of the modified Slip Serial with this library is here: https://github.com/miwos/miwos/blob/main/bridge/device/lib/Bridge/include/Bridge.h#L121

fcana, 2023-12-09:
> @fcana I'm using this library with a Teensy 4.1 and a slightly [modified Slip Serial](https://github.com/miwos/miwos/blob/main/bridge/firmware/lib/Bridge/include/SlipSerial.h) Usage of the modified [...]

thank you very much @arnoson for that... I'll try using that in the next days!

fcana, 2023-12-09:
> @fcana I'm using this library with a Teensy 4.1 and a slightly [modified Slip Serial](https://github.com/miwos/miwos/blob/main/bridge/firmware/lib/Bridge/include/SlipSerial.h) Usage of the modified [...]

thank you @arnoson but it is not working for me... i need SlipEncoded*USB*Serial in my project and I can't figure out how to use your mod library... any advice?

arnoson, 2023-12-09:
This should work with a usb serial, at least I pass the `SlipSerial` class a reference of type `usb_serial_class` in the constructor or am I missing something?