CNMAT/OSC #116: using integers instead of string addresses

traviscweller, 2020-10-22:
I'm attempting to use this library to accept OSC generated by supercollider, which uses an integer substitution for the address portion of the OSC message to allow more efficient processing by the synth server. For example:
```
b.sendMsg("/s_new", "inst", 1001); // gets translated into:
b.sendMsg(9, "inst", 1001);

```
You can see the normal command address "/s_new" is sent as "9" - a 32 bit integer - instead of an OSC style /string. 
I recognize this is non-standard OSC syntax, but it is also a commonly used application.

Is there any possibility of adding support for 32bit integer style addresses to dispatch? Or maybe I'm missing something and this is already possible with a work-around?

adrianfreed, 2020-10-23:
Thanks for your interest.  Neither CNMAT or the original authors actively develop this library, so it
may be easier to find someone able to add this features in the supercollider community.
I will keep an eye out for a pull request and seriously consider including it if someone does work on it.

traviscweller, 2020-10-24:
Thanks Adrian!

I've reached out to the supercollider mailing list to see if there is any
interest. I'd do it myself, but having looked at the code -- it would be an
uphill battle given my skill set. I appreciate the reply and your
consideration!

All best,
Travis

closed 2024-05-13 by adrianfreed