CNMAT/OSC #121: ESP32 OSC over TCP?

thorangutang, 2021-04-23:
Hi there, I have been using the OSC library with UDP to send and receive bundles from ESP32's for a good while and it mostly works really well. 

However, I do at times experience dropped packages and since this is for  robotic applications where I am generally sending commands and not continuous data, I was wondering if there are any examples as to how to use TCP as the transport. The highest priority is for all messages to actually arrive(in order).

Any pointers?

Thank you!

adrianfreed, 2021-04-25:
Thanks for your interest.

TCP connections - when established provide a transport for streams of bytes. When you set up tutorials for that with the ESP32 and confirm
you have a satisfactory stream for your application, I recommend you use slip encoded OSC packets set up in the library demos
for serial streams like UARTs and USB - they will work fine over TCP too.

fastfourier666, 2022-10-20:
I did try using this library with TCP but can't find a way to use the SLIPEncodedSerial class - it always expects a HardwareSerial as a constructor.

I ended up using the PacketSerial library to do the SLIP decoding. It allows you to set the stream to the WiFiClient. When a SLIP packet is completed you get a callback, and from there you can do a `msg.fill` with the received bytes. The sending end is the osc.js library.

I did have a try at making my own "SLIPEncodedTCP" using the "SLIPEncodedSerial" class as a starting point but ran out of time. OSC over TCP is great, especially with these little wifi boards with small antennae that can often get a crappy signal.

fastfourier666, 2022-10-21:
I went back to it and made it work. [example](https://github.com/fastfourier666/CNMAT_OSC_TCP_EXAMPLE)

The SLIPEncodedTCP class I made only handles sending and receiving. You have to manage the client's open/closed state out in the sketch.

I wonder if this is all unnecessary ---- but I couldn't find any way of getting the SLIPEncodedSerial or SLIPEncodedUSBSerial to work as-is. You can recast to HardwareSerial when creating the object but the class calls methods on it like `begin()`, which don't work on a WiFiClient.