CNMAT/OSC #123: I want to forward OSC to more than one IP address.

taehyeops, 2021-07-12:
Is there any way to forward the same OSC command to two IP addresses?

I am using a playback application called "Qlab", which is used in musicals, running on two laptops. 

In the event of a problem on one laptop, a spare system is configured and used so that the other laptop can play it.

At this time, I want to send a command to both laptops using OSC to control them.

Currently, sending to one laptop through the provided OSC library is successful. Very easy and good library. thanks..

I wonder if the above is possible.

JonasVorwerk, 2021-07-13:
Hi,

You can try using ip address 255.255.255.255, this will send it to all connected devices.

All the best,

Jonas

adrianfreed, 2021-07-13:
Thanks for your interest in the library.
Here is an old post discussing the scenario you are interested in:
https://forum.arduino.cc/t/udp-packets-to-multiple-ips/305326

That post expresses concerns about delays. The Arduino ethernet library has evolved a lot since
then so it is worth checking that now.

closed 2024-05-13 by adrianfreed