CNMAT/OSC #124: message lenght from max msp

andreatrona, 2021-07-16:
Hi all, 
I have an issue that I don't understand: I have to send a text that is in a json file through OSC from MaxMsp to Teensy to save it on SD card... but when the json file reach a size that i don't know the message is no longer received.
Some advice about OSCMessage max size? Thanks

adrianfreed, 2021-07-17:
What kind of Teensy are you using? Are you using serial over USB or ethernet?
We dynamically allocate memory on the Teensy so there may be a failure there depending on which model of Teensy.
The teensy IDE reports available memory when you finish compiling

andreatrona, 2021-07-17:
Hi Adrian and thank you for your answer.
I'm using a Teensy 3.5 and ethernet via Adafruit Ethernet FeatherWing. 
This is the report when I finish compiling: 

Sketch uses 54564 bytes (10%) of program storage space. Maximum is 524288 bytes.
Global variables use 11424 bytes (4%) of dynamic memory, leaving 250712 bytes for local variables. Maximum is 262136 bytes.

For a more clear explanation of what I'm doing.
I send from max msp a char array (a json file serialized) to store it in an sd card. After stored, they can be read from a library that menage json object...
So, when the char array that I send from max msp become long, it stop working. 
Here the text that I want to send. This is the length reached that
`{
	"faro0" : 	{
		"red" : 1,
		"green" : 2,
		"blue" : 3,
		"dimmer" : 4,
		"strobo" : 5,
		"pan" : 6,
		"tilt" : 7
	}
,
	"faro1" : 	{
		"red" : 11,
		"green" : 12,
		"blue" : 13,
		"dimmer" : 14,
		"strobo" : 15,
		"pan" : 16,
		"tilt" : 17
	}
,
	"faro2" : 	{
		"red" : 21,
		"green" : 22,
		"blue" : 23,
		"dimmer" : 34,
		"strobo" : 25,
		"pan" : 26,
		"tilt" : 27
	}
,
	"faro3" : 	{
		"red" : 31,
		"green" : 32,
		"blue" : 33,
		"dimmer" : 34,
		"strobo" : 35,
		"pan" : 36,
		"tilt" : 37
	}
,
	"faro4" : 	{
		"red" : 41,
		"green" : 42,
		"blue" : 43,
		"dimmer" : 44,
		"strobo" : 45,
		"pan" : 46,
		"tilt" : 47
	}
,
	"faro5" : 	{
		"red" : 51,
		"green" : 52,
		"blue" : 53,
		"dimmer" : 54,
		"strobo" : 55,
		"pan" : 56,
		"tilt" : 57
	}
,
	"faro6" : 	{
		"red" : 61,
		"green" : 62,
		"blue" : 63,
		"dimmer" : 64,
		"strobo" : 65,
		"pan" : 66,
		"tilt" : 67
	}
,
	"faro7" : 	{
		"red" : 71,
		"green" : 72,
		"blue" : 73,
		"dimmer" : 74,
		"strobo" : 75,
		"pan" : 76,
		"tilt" : 77
	}
,
	"faro8" : 	{
		"red" : 81,
		"green" : 82,
		"blue" : 83,
		"dimmer" : 84,
		"strobo" : 85,
		"pan" : 86,
		"tilt" : 87
	}
,
	"faro9" : 	{
		"red" : 91,
		"green" : 92,
		"blue" : 93,
		"dimmer" : 94,
		"strobo" : 95,
		"pan" : 96,
		"tilt" : 97
	}
,
	"faro10" : 	{
		"red" : 1,
		"green" : 2,
		"blue" : 3,
		"dimmer" : 4,
		"strobo" : 5,
		"pan" : 6,
		"tilt" : 7
	}
,
	"faro11" : 	{
		"red" : 1,
		"green" : 2
	}

}
`

andreatrona, 2021-07-17:
I also closed the issue wrongly, let's reopen it

andreatrona, 2021-07-19:
Ok, I'm taking another way to do this. 
I'm trying to send nested osc message that correspond to the dictionary in max (that read a json file format); for every data that I'll receive on the teensy via osc, I will start building the json document and then, when I received all of the data I'll serialize the json document to the SD card.
Hope it will work, I'll tell you if I have other issue.

thanks Adrian

adrianfreed, 2021-07-19:
There is a limit called the MTU established by the 
W5500 in the ethernet Featherwing.
It’s usually 1 or 2 thousand bytes.

Your data seems smaller than that so I am still mystified.
Perhaps you can share the code so I can look reproduce.

andreatrona, 2021-07-26:
Hi Adrian, I found the limit called MTU, it's about 1472 bytes for udp. When in my sketch i print the size of the incoming message, it's around this number.

Now I'm using a system that interpret and read osc tags and "route" the message in the right path.

Thanks for your replies and for developing this tools.
Andrea

andreatrona, 2021-07-26:
It's useful for me to understand the path because I need to create a json document with nested object through ArduinoJson library

bye

closed 2024-05-13 by adrianfreed