CNMAT/OSC #136: Missing argument types: m, h, r, N, I

wanthalf, 2022-10-24:
Any reason why the rest of extended argument types is not supported? I mean especially the m[idi], but eventually also h [int64], r[gba], N[il] and I[nfinitum]. The library seems to be designed very well, so that it should only be question of adding the appropriate handlers. I guess that only arrays could be more difficult to deal with.

I am also a bit confused whether c[har] (and T[rue], [F]alse?) is fully implemented both for reception and sending, or not. It is covered by some methods but seems to be missing in others (OSCMessage::decodeData covers i,f,d,t,s,b, but not c,T,F).

adrianfreed, 2024-05-04:
The original developers moved to other projects and CNMAT is no longer working on this.
I will look at PRs

adrianfreed, 2024-05-29:
I added rgba.
Will be adding midi next.