CNMAT/OSC #139: 2023 Updates - broken BluetoothSerial with pio and esp32

redFrik, 2023-09-05:
Thanks for the new release. Could it be that there's a dependency missing?

```
% pio run
Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
----------------------------------------------------------------------------------------------------
Tool Manager: Installing platformio/framework-arduinoespressif32 @ ~3.20011.0
Unpacking  [####################################]  100%
Tool Manager: framework-arduinoespressif32@3.20011.230801 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.4.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20011.230801 (2.0.11)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 40 compatible libraries
Scanning dependencies...
Dependency Graph
|-- OSC @ 1.0.0
|-- WiFi @ 2.0.0
Building in release mode
Compiling .pio/build/esp32dev/src/main.cpp.o
Building .pio/build/esp32dev/bootloader.bin
Generating partitions .pio/build/esp32dev/partitions.bin
esptool.py v4.5.1
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Compiling .pio/build/esp32dev/libb6c/OSC/OSCBoards.cpp.o
Compiling .pio/build/esp32dev/libb6c/OSC/OSCBundle.cpp.o
Compiling .pio/build/esp32dev/libb6c/OSC/OSCData.cpp.o
Compiling .pio/build/esp32dev/libb6c/OSC/OSCMatch.c.o
Compiling .pio/build/esp32dev/libb6c/OSC/OSCMessage.cpp.o
Compiling .pio/build/esp32dev/libb6c/OSC/OSCTiming.cpp.o
Compiling .pio/build/esp32dev/libb6c/OSC/SLIPEncodedBluetoothSerial.cpp.o
Compiling .pio/build/esp32dev/libb6c/OSC/SLIPEncodedSerial.cpp.o
In file included from /Users/xyz/.platformio/lib/OSC/SLIPEncodedBluetoothSerial.cpp:1:
/Users/xyz/.platformio/lib/OSC/SLIPEncodedBluetoothSerial.h:11:10: fatal error: BluetoothSerial.h: No such file or directory
```

This on macOS with platformio installed from homebrew.

Your previous release (3.5.7) works fine.

rahji, 2024-01-11:
This error is still occurring. The workaround is to remove the two slipbluetooth header and source files from the PIO libdeps directory. I thought it was because I was using an esp8266 with no Bluetooth but it's happening with esp32 as well.

hughqelliott, 2024-02-13:
Any resolution on this? I only started using the library to find this issue.

bmentink, 2024-03-04:
I am getting this error as well ..
```
Compiling .pio/build/lolin_s2_mini/libd63/OSC/SLIPEncodedBluetoothSerial.cpp.o
Compiling .pio/build/lolin_s2_mini/libd63/OSC/SLIPEncodedSerial.cpp.o
Compiling .pio/build/lolin_s2_mini/libd63/OSC/SLIPEncodedUSBSerial.cpp.o
In file included from .pio/libdeps/lolin_s2_mini/OSC/SLIPEncodedBluetoothSerial.cpp:1:
.pio/libdeps/lolin_s2_mini/OSC/SLIPEncodedBluetoothSerial.h:20:2: error: 'BluetoothSerial' does not name a type
  BluetoothSerial * serial;
  ^~~~~~~~~~~~~~~
.pio/libdeps/lolin_s2_mini/OSC/SLIPEncodedBluetoothSerial.h:26:44: error: expected ')' before '&' token
  SLIPEncodedBluetoothSerial(BluetoothSerial & );
                            ~               ^~
                                            )
.pio/libdeps/lolin_s2_mini/OSC/SLIPEncodedBluetoothSerial.cpp:9:55: error: expected constructor, destructor, or type conversion before '(' token
 SLIPEncodedBluetoothSerial::SLIPEncodedBluetoothSerial(BluetoothSerial &s){
                                                       ^
.pio/libdeps/lolin_s2_mini/OSC/SLIPEncodedBluetoothSerial.cpp: In member function 'bool SLIPEncodedBluetoothSerial::endofPacket()':
```

bmentink, 2024-03-04:
However,  when I run the code It connects to Wifi ok, but in `loop()` I get an error even before I have sent any OSC data.
The contents of loop() is:
```
void loop() {
  // OSC Packet handling
  int size = Udp.parsePacket();
	while (size--) {
	  bundleIN.fill(Udp.read());
	}
	if (!bundleIN.hasError()) {
	  bundleIN.dispatch("/1/slider2", motorControl);
	    IPAddress ip = Udp.remoteIP();
	    if (!( ip==outIp )) {
	      Serial.print("New connection from ");
	      Serial.println(ip);
	      outIp = ip;
	    }
  }
	 else {
	  error = bundleIN.getError();
	  Serial.print("error: ");
	  Serial.println(error);
	}
	bundleIN.empty();
}
```

closed 2024-05-13 by adrianfreed

rahji, 2024-05-13:
Is this closed because my workaround of deleting the files seems to work or has it been fixed?

pingarelho, 2024-06-23:
I'm having this same issue.

f0x52, 2024-09-27:
This is indeed still a problem with ESP32, maybe reopen the issue?

f0x52, 2024-10-26:
Update: this was already fixed in the repo, but the platformio registry (still) hasn't been updated. If you add `https://github.com/cnmat/OSC` as a dependency instead of `cnmat/OSC`, it works.

Fixed in https://github.com/CNMAT/OSC/commit/a94dc84f7b6a60c08db9f8b72aabedda6daf1354