CNMAT/OSC #144: compile for esp32 fails

trackme518, 2024-01-07:
Hi,
when using esp32 definitions version 2.0.14 (https://github.com/espressif/arduino-esp32) I can compile for esp32 s2 mini but not for esp32 c3 super mini. The code is unchanged for these two. Errors below. Also when I tried to upgrade the board definitions for esp32 to to 3.0.0 alpha3 it will fail for esp32 s2 mini as well. 
```
In file included from c:\Users\neuromancer\Documents\Arduino\libraries\OSC-3.5.8/OSCBundle.h:29,
                 from C:\Users\neuromancer\Documents\GitHub\datainmotion\ArduinoIDE\moduleWifiSerial\OscManager.hpp:4,
                 from C:\Users\neuromancer\Documents\GitHub\datainmotion\ArduinoIDE\moduleWifiSerial\moduleWifiSerial.ino:14:
c:\Users\neuromancer\Documents\Arduino\libraries\OSC-3.5.8/OSCMessage.h: In instantiation of 'OSCMessage& OSCMessage::add(T) [with T = int]':
C:\Users\neuromancer\Documents\GitHub\datainmotion\ArduinoIDE\moduleWifiSerial\OscManager.hpp:116:39:   required from here
c:\Users\neuromancer\Documents\Arduino\libraries\OSC-3.5.8/OSCMessage.h:136:17: error: call of overloaded 'OSCData(int&)' is ambiguous
   OSCData * d = new OSCData(datum);
                 ^~~~~~~~~~~~~~~~~~
In file included from c:\Users\neuromancer\Documents\Arduino\libraries\OSC-3.5.8/OSCMessage.h:29,
                 from c:\Users\neuromancer\Documents\Arduino\libraries\OSC-3.5.8/OSCBundle.h:29,
                 from C:\Users\neuromancer\Documents\GitHub\datainmotion\ArduinoIDE\moduleWifiSerial\OscManager.hpp:4,
                 from C:\Users\neuromancer\Documents\GitHub\datainmotion\ArduinoIDE\moduleWifiSerial\moduleWifiSerial.ino:14:
c:\Users\neuromancer\Documents\Arduino\libraries\OSC-3.5.8/OSCData.h:106:5: note: candidate: 'OSCData::OSCData(boolean)'
     OSCData  (boolean);
     ^~~~~~~
c:\Users\neuromancer\Documents\Arduino\libraries\OSC-3.5.8/OSCData.h:102:2: note: candidate: 'OSCData::OSCData(double)'
  OSCData (double);
  ^~~~~~~
c:\Users\neuromancer\Documents\Arduino\libraries\OSC-3.5.8/OSCData.h:101:2: note: candidate: 'OSCData::OSCData(float)'
  OSCData (float);
  ^~~~~~~
c:\Users\neuromancer\Documents\Arduino\libraries\OSC-3.5.8/OSCData.h:100:5: note: candidate: 'OSCData::OSCData(unsigned int)'
     OSCData (unsigned int);
     ^~~~~~~
c:\Users\neuromancer\Documents\Arduino\libraries\OSC-3.5.8/OSCData.h:96:2: note: candidate: 'OSCData::OSCData(int32_t)'
  OSCData (int32_t);
  ^~~~~~~
c:\Users\neuromancer\Documents\Arduino\libraries\OSC-3.5.8/OSCData.h:67:5: note: candidate: 'OSCData::OSCData(char)'
     OSCData(char t);
     ^~~~~~~
c:\Users\neuromancer\Documents\Arduino\libraries\OSC-3.5.8/OSCData.h:57:7: note: candidate: 'constexpr OSCData::OSCData(const OSCData&)'
 class OSCData
       ^~~~~~~
In file included from c:\Users\neuromancer\Documents\Arduino\libraries\OSC-3.5.8/OSCBundle.h:29,
                 from C:\Users\neuromancer\Documents\GitHub\datainmotion\ArduinoIDE\moduleWifiSerial\OscManager.hpp:4,
                 from C:\Users\neuromancer\Documents\GitHub\datainmotion\ArduinoIDE\moduleWifiSerial\moduleWifiSerial.ino:14:
c:\Users\neuromancer\Documents\Arduino\libraries\OSC-3.5.8/OSCMessage.h: In instantiation of 'OSCMessage& OSCMessage::add(T) [with T = short int]':
C:\Users\neuromancer\Documents\GitHub\datainmotion\ArduinoIDE\moduleWifiSerial\moduleWifiSerial.ino:65:37:   required from here
c:\Users\neuromancer\Documents\Arduino\libraries\OSC-3.5.8/OSCMessage.h:136:17: error: call of overloaded 'OSCData(short int&)' is ambiguous
   OSCData * d = new OSCData(datum);
                 ^~~~~~~~~~~~~~~~~~
In file included from c:\Users\neuromancer\Documents\Arduino\libraries\OSC-3.5.8/OSCMessage.h:29,
                 from c:\Users\neuromancer\Documents\Arduino\libraries\OSC-3.5.8/OSCBundle.h:29,
                 from C:\Users\neuromancer\Documents\GitHub\datainmotion\ArduinoIDE\moduleWifiSerial\OscManager.hpp:4,
                 from C:\Users\neuromancer\Documents\GitHub\datainmotion\ArduinoIDE\moduleWifiSerial\moduleWifiSerial.ino:14:
c:\Users\neuromancer\Documents\Arduino\libraries\OSC-3.5.8/OSCData.h:106:5: note: candidate: 'OSCData::OSCData(boolean)'
     OSCData  (boolean);
     ^~~~~~~
c:\Users\neuromancer\Documents\Arduino\libraries\OSC-3.5.8/OSCData.h:102:2: note: candidate: 'OSCData::OSCData(double)'
  OSCData (double);
  ^~~~~~~
c:\Users\neuromancer\Documents\Arduino\libraries\OSC-3.5.8/OSCData.h:101:2: note: candidate: 'OSCData::OSCData(float)'
  OSCData (float);
  ^~~~~~~
c:\Users\neuromancer\Documents\Arduino\libraries\OSC-3.5.8/OSCData.h:100:5: note: candidate: 'OSCData::OSCData(unsigned int)'
     OSCData (unsigned int);
     ^~~~~~~
c:\Users\neuromancer\Documents\Arduino\libraries\OSC-3.5.8/OSCData.h:96:2: note: candidate: 'OSCData::OSCData(int32_t)'
  OSCData (int32_t);
  ^~~~~~~
c:\Users\neuromancer\Documents\Arduino\libraries\OSC-3.5.8/OSCData.h:67:5: note: candidate: 'OSCData::OSCData(char)'
     OSCData(char t);
     ^~~~~~~
c:\Users\neuromancer\Documents\Arduino\libraries\OSC-3.5.8/OSCData.h:57:7: note: candidate: 'constexpr OSCData::OSCData(const OSCData&)'
 class OSCData
       ^~~~~~~
```

trackme518, 2024-01-07:
Hi,
the problem was that library somehow does not account for int / int16 when using esp32 c3? I have solved it by casting all integers to 32bit like `message.add( (int32_t) int-variable ); `

trackme518, 2024-01-07:
Maybe another type can be added to constructor with int16_t type?

adrianfreed, 2024-05-04:
Your solution is the correct one. If we provided a an add that worked for int16_t the design would call for it to use a special osc type for 16 bit integers which is not standard. People have done it though.

I am toying with the idea of adding a method:
addAsOSCint()

This would avoid the cast and the need to bake in the OSC integer format type.

adrianfreed, 2024-05-08:
I addressed this and all sorts of issues due to compiler variations in the latest Main.
Recommendation is to use intOSC_t as the cast. I have tested on a lot of platforms. Not quite all of them yet though.

closed 2024-05-13 by adrianfreed