CNMAT/OSC #148: ESP32-C3: Error: "no declaration matches 'OSCData::OSCData(uint32_t)"

ritsch, 2024-05-07:
Compiling within a platformio project on ESP32-C3, the master branch of CNMAT/OSC: 

`platformio.ini:` `libdeps =  https://github.com/CNMAT/OSC`  
loads `(OSC @ 3.5.8+sha.7912a02)`

Trigger an compile error see below:

going back to version: 
`libdeps = https://github.com/CNMAT/OSC#3.5.8`
works

````
.pio/libdeps/serial-upload/OSC/OSCData.cpp:37:1: error: no declaration matches 'OSCData::OSCData(uint32_t)'
 OSCData::OSCData(uint32_t i){
 ^~~~~~~
In file included from .pio/libdeps/serial-upload/OSC/OSCData.cpp:2:
.pio/libdeps/serial-upload/OSC/OSCData.h:57:7: note: candidates are: 'constexpr OSCData::OSCData(const OSCData&)'
 class OSCData
       ^~~~~~~
.pio/libdeps/serial-upload/OSC/OSCData.h:128:2: note:                 'OSCData::OSCData(char, uint8_t*, int)'
  OSCData(char, uint8_t *, int);
  ^~~~~~~
.pio/libdeps/serial-upload/OSC/OSCData.h:107:5: note:                 'OSCData::OSCData(osctime_t)'
     OSCData  (osctime_t);
     ^~~~~~~
.pio/libdeps/serial-upload/OSC/OSCData.h:106:5: note:                 'OSCData::OSCData(boolean)'
     OSCData  (boolean);
     ^~~~~~~
.pio/libdeps/serial-upload/OSC/OSCData.h:105:2: note:                 'OSCData::OSCData(OSCData*)'
  OSCData (OSCData *);
  ^~~~~~~
.pio/libdeps/serial-upload/OSC/OSCData.h:103:2: note:                 'OSCData::OSCData(uint8_t*, int)'
  OSCData (uint8_t *, int);
  ^~~~~~~
.pio/libdeps/serial-upload/OSC/OSCData.h:102:2: note:                 'OSCData::OSCData(double)'
  OSCData (double);
  ^~~~~~~
.pio/libdeps/serial-upload/OSC/OSCData.h:101:2: note:                 'OSCData::OSCData(float)'
  OSCData (float);
  ^~~~~~~
.pio/libdeps/serial-upload/OSC/OSCData.h:100:5: note:                 'OSCData::OSCData(unsigned int)'
     OSCData (unsigned int);
     ^~~~~~~
.pio/libdeps/serial-upload/OSC/OSCData.cpp:31:1: note:                 'OSCData::OSCData(int32_t)'
 OSCData::OSCData(int32_t i){
 ^~~~~~~
.pio/libdeps/serial-upload/OSC/OSCData.cpp:14:1: note:                 'OSCData::OSCData(const char*)'
 OSCData::OSCData(const char * s){
 ^~~~~~~
In file included from .pio/libdeps/serial-upload/OSC/OSCData.cpp:2:
.pio/libdeps/serial-upload/OSC/OSCData.h:67:5: note:                 'OSCData::OSCData(char)'
     OSCData(char t);
     ^~~~~~~
.pio/libdeps/serial-upload/OSC/OSCData.h:57:7: note: 'class OSCData' defined here
 class OSCData
       ^~~~~~~
````

Something went wrong fixing the previous issue. 

PS.: For information:  used `https://github.com/CNMAT/OSC` since  `lib_deps = CNMAT/OSC` throw the error: 
```
In file included from .pio/libdeps/serial-upload/OSC/SLIPEncodedBluetoothSerial.cpp:1:
.pio/libdeps/serial-upload/OSC/SLIPEncodedBluetoothSerial.h:11:10: fatal error: BluetoothSerial.h: No such file or directory

```
since ESP32-C3 has no bluetooth.

mfG
  Winfried

adrianfreed, 2024-05-08:
Thanks
I just pushed a better handling of types

ritsch, 2024-05-08:
Thank you for the fast response,
tested and the error is closed, 

... but one more error is opened:

```
.pio/libdeps/serial-upload/OSC/OSCMessage.h: In instantiation of 'OSCMessage& OSCMessage::add(T) [with T = long int]':
src/pl_main.cpp:98:31:   required from here
.pio/libdeps/serial-upload/OSC/OSCMessage.h:136:17: error: call of overloaded 'OSCData(long int&)' is ambiguous
   OSCData * d = new OSCData(datum);
                 ^~~~~~~~~~~~~~~~~~
In file included from .pio/libdeps/serial-upload/OSC/OSCMessage.h:29,
                 from .pio/libdeps/serial-upload/OSC_control/src/osc_control.h:40,
                 from .pio/libdeps/serial-upload/LEMiotLib/src/lemiot.h:21,
                 from src/pl_main.cpp:17:
.pio/libdeps/serial-upload/OSC/OSCData.h:113:5: note: candidate: 'OSCData::OSCData(boolean)'
     OSCData  (boolean);
     ^~~~~~~
.pio/libdeps/serial-upload/OSC/OSCData.h:109:2: note: candidate: 'OSCData::OSCData(double)'
  OSCData (double);
  ^~~~~~~
.pio/libdeps/serial-upload/OSC/OSCData.h:108:2: note: candidate: 'OSCData::OSCData(float)'
  OSCData (float);
  ^~~~~~~
.pio/libdeps/serial-upload/OSC/OSCData.h:107:5: note: candidate: 'OSCData::OSCData(unsigned int)'
     OSCData (unsigned int);
     ^~~~~~~
.pio/libdeps/serial-upload/OSC/OSCData.h:103:3: note: candidate: 'OSCData::OSCData(intOSC_t)'
   OSCData (intOSC_t);
   ^~~~~~~
.pio/libdeps/serial-upload/OSC/OSCData.h:72:5: note: candidate: 'OSCData::OSCData(char)'
     OSCData(char t);
     ^~~~~~~
.pio/libdeps/serial-upload/OSC/OSCData.h:62:7: note: candidate: 'constexpr OSCData::OSCData(const OSCData&)'
 class OSCData
       ^~~~~~~

```

Dont know if this is specific to the ESP32-C3 or also for others.

mfG Winfried

adrianfreed, 2024-05-08:
thanks. I will take a look at that one too. So many compiler variants now.
Some vendors like m5stack - one of my favorites - are behind in bug fixes in esp32 dev environment.

adrianfreed, 2024-05-08:
Can you give me the Board name please?
I have other esp32-c3 boards working

ritsch, 2024-05-09:
Use mostly olimex boards, since real open hardware: ESP32-C3-devkit-LiPo Rev.A, since there is no dedicated board in platformio, I use the compatible board in `platformio.ini`: 

`board = esp32-c3-devkitm-1`

see compile output:

```
...
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-c3-devkitm-1.html
PLATFORM: Espressif 32 (6.4.0) > Espressif ESP32-C3-DevKitM-1
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-builtin) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
...
```

If it works with you, I can do some further investigations.

adrianfreed, 2024-05-09:
I changed things recently to address various compiler differences. Now intOSC_t should work for all platforms. Sometimes it is int32_t other times it will be int...

ritsch, 2024-05-12:
Thanks for the hint. There was some sloppy typedefs of `int32_t` in my libs [1,2], which I cast now to `intOSC_t` and now it compiles, so no error anymore.

So I think the issue is solved.

mfg winfried

[1] https://git.iem.at/uC/LEMiotLib.git
[2] https://git.iem.at/uC/OSC_networking.git

closed 2024-05-13 by adrianfreed