CNMAT/OSC #151: ESP8266 OSCData(int) cannot be overloaded with OSCData(intOSC_t)

clizzy2, 2024-08-15:
Hi!

I'm very new to arduino, and C++, but I can't verify any of the example files of the ESP8266 because I get the following error

In file included from v:\Arduino\libraries\OSC\OSCMessage.h:29,
                 from v:\Arduino\libraries\OSC\OSCBundle.h:29,
                 from v:\Arduino\libraries\OSC\OSCBundle.cpp:26:
v:\Arduino\libraries\OSC\OSCData.h:124:5: error: 'OSCData::OSCData(int)' cannot be overloaded with 'OSCData::OSCData(intOSC_t)'
  124 |     OSCData (int);
      |     ^~~~~~~
v:\Arduino\libraries\OSC\OSCData.h:121:3: note: previous declaration 'OSCData::OSCData(intOSC_t)'
  121 |   OSCData (intOSC_t);
      |   ^~~~~~~
In file included from v:\Arduino\libraries\OSC\OSCData.cpp:2:
v:\Arduino\libraries\OSC\OSCData.h:124:5: error: 'OSCData::OSCData(int)' cannot be overloaded with 'OSCData::OSCData(intOSC_t)'
  124 |     OSCData (int);
      |     ^~~~~~~
v:\Arduino\libraries\OSC\OSCData.h:121:3: note: previous declaration 'OSCData::OSCData(intOSC_t)'
  121 |   OSCData (intOSC_t);
      |   ^~~~~~~
v:\Arduino\libraries\OSC\OSCData.cpp:42:1: error: redefinition of 'OSCData::OSCData(int)'
   42 | OSCData::OSCData(int i){
      | ^~~~~~~
v:\Arduino\libraries\OSC\OSCData.cpp:34:1: note: 'OSCData::OSCData(intOSC_t)' previously defined here
   34 | OSCData::OSCData(intOSC_t i){
      | ^~~~~~~
In file included from v:\Arduino\libraries\OSC\OSCMessage.h:29,
                 from v:\Arduino\libraries\OSC\OSCMessage.cpp:26:
v:\Arduino\libraries\OSC\OSCData.h:124:5: error: 'OSCData::OSCData(int)' cannot be overloaded with 'OSCData::OSCData(intOSC_t)'
  124 |     OSCData (int);
      |     ^~~~~~~
v:\Arduino\libraries\OSC\OSCData.h:121:3: note: previous declaration 'OSCData::OSCData(intOSC_t)'
  121 |   OSCData (intOSC_t);
      |   ^~~~~~~

exit status 1

Compilation error: exit status 1

I'm running Arduino IDE 2.3.2. Should I be running 1.8.5?

julienrobert, 2024-08-23:
Hi, I have the same issue. I'm on Mac 14.6.1 Apple Silicon M2.

I tested with 1.8.19 and 2.3.2. 1.5.8 won't run on my system.

TyIsI, 2024-08-29:
Version 3.5.8 should work as intended:

https://github.com/CNMAT/OSC/releases/tag/3.5.8

@adrianfreed It looks like https://github.com/CNMAT/OSC/commit/a1fc01215012d6128b701ccb13d6e17c5c4b0397 introduced a regression?

julienrobert, 2024-09-08:
I'm still having the issue.

TyIsI, 2024-09-09:
@julienrobert which version of the OSC library are you using?

julienrobert, 2024-09-09:
I tried the latest and 3.5.8.

zealtv, 2024-10-10:
Same issue here. 

Tested on:
- Mac Silicon 14.6.1
- Arduino 2.3.3 and 1.8.19
- OSC library versions 3.5.8, 1.35, and 1.37.  

Trying to build [this sketch](https://github.com/playablestreets/plantsOS/blob/13e8e3547a86fed7788141e1b3dbafa56952356c/arduino/esposc_mpr121/esposc_mpr121.ino) which has built happily in the past.  No idea what has changed...

LoicLac, 2024-11-12:
I have the same issue here , 
apple silicon M1 14.1.1
Arduino 2.3.3 and 1.8.19
OSC library versions 3.5.8, 1.35, and 1.37.

my sketch use to build ...

rvendrame, 2025-07-11:
Same issue here,  iMac Intel,  Macbook,  Arduino 2.3.6,  OSC 3.5.8.
Did anyone found a fix for this?