CNMAT/OSC #154: OSCData.h wired issues with adding `int32_t` data

sstaub, 2024-10-16:
Adding `int32_t` data doesn't work on some microcontrollers.
It works on an Arduino UNO but when trying to compile Raspberry Pi Pico I get some error message.
`int` or `int64_t` works as expected.
The problem comes up with last update of OSCData.h

```
/Users/sstaub/Documents/Arduino/libraries/OSC/OSCMessage.h: In instantiation of 'OSCMessage& OSCMessage::add(T) [with T = long int]':
/Users/sstaub/Documents/Arduino/ETC_EOS_TCP/ETC_EOS_TCP.ino:50:11:   required from here
   50 |                 ping.add(pingNum++);
      |                 ~~~~~~~~^~~~~~~~~~~
/Users/sstaub/Documents/Arduino/libraries/OSC/OSCMessage.h:136:31: error: call of overloaded 'OSCData(long int&)' is ambiguous
  136 |                 OSCData * d = new OSCData(datum);
      |                               ^~~~~~~~~~~~~~~~~~
In file included from /Users/sstaub/Documents/Arduino/libraries/OSC/OSCMessage.h:29:
/Users/sstaub/Documents/Arduino/libraries/OSC/OSCData.h:132:5: note: candidate: 'OSCData::OSCData(boolean)'
  132 |     OSCData  (boolean);
      |     ^~~~~~~
/Users/sstaub/Documents/Arduino/libraries/OSC/OSCData.h:128:9: note: candidate: 'OSCData::OSCData(double)'
  128 |         OSCData (double);
      |         ^~~~~~~
/Users/sstaub/Documents/Arduino/libraries/OSC/OSCData.h:127:9: note: candidate: 'OSCData::OSCData(float)'
  127 |         OSCData (float);
      |         ^~~~~~~
/Users/sstaub/Documents/Arduino/libraries/OSC/OSCData.h:126:5: note: candidate: 'OSCData::OSCData(unsigned int)'
  126 |     OSCData (unsigned int);
      |     ^~~~~~~
/Users/sstaub/Documents/Arduino/libraries/OSC/OSCData.h:122:10: note: candidate: 'OSCData::OSCData(int64_t)'
  122 |          OSCData (int64_t);
      |          ^~~~~~~
/Users/sstaub/Documents/Arduino/libraries/OSC/OSCData.h:121:10: note: candidate: 'OSCData::OSCData(intOSC_t)'
  121 |          OSCData (intOSC_t);
      |          ^~~~~~~
/Users/sstaub/Documents/Arduino/libraries/OSC/OSCData.h:87:5: note: candidate: 'OSCData::OSCData(char)'
   87 |     OSCData(char t);
      |     ^~~~~~~
/Users/sstaub/Documents/Arduino/libraries/OSC/OSCData.h:77:7: note: candidate: 'constexpr OSCData::OSCData(const OSCData&)'
   77 | class OSCData
      |       ^~~~~~~

Using library Ethernet at version 2.0.2 in folder: /Users/sstaub/Documents/Arduino/libraries/Ethernet 
Using library OSC at version 3.5.8 in folder: /Users/sstaub/Documents/Arduino/libraries/OSC 
Using library SPI at version 1.0 in folder: /Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/4.1.1/libraries/SPI 
exit status 1

Compilation error: exit status 1
```

adrianfreed, 2024-12-02:
Does it compile when you cast pingNum++ to intOSC_t
?

adrianfreed, 2024-12-02:
That IDE may be using a more recent gcc which uses more stringent type equivalence rules

sstaub, 2024-12-02:
I tested it with Arduino IDE. But I can't say which compiler version is used by RaspberryPiPico.

sstaub, 2024-12-31:
An additional info, using the latest Arduino IDE on Windows, int32_t works without a problem.

rvendrame, 2025-07-23:
Interesting, it works fine for ESP32,  however with ESP8266,  the int x intOSC_t still there: 

I```
n file included from /Users/rvendrame/Documents/Arduino/libraries/OSC/OSCData.cpp:2:
/Users/rvendrame/Documents/Arduino/libraries/OSC/OSCData.h:124:5: error: 'OSCData::OSCData(int)' cannot be overloaded with 'OSCData::OSCData(intOSC_t)'
  124 |     OSCData (int);
      |     ^~~~~~~
/Users/rvendrame/Documents/Arduino/libraries/OSC/OSCData.h:121:3: note: previous declaration 'OSCData::OSCData(intOSC_t)'
  121 |   OSCData (intOSC_t);
      |   ^~~~~~~
/Users/rvendrame/Documents/Arduino/libraries/OSC/OSCData.cpp:42:1: error: redefinition of 'OSCData::OSCData(int)'
   42 | OSCData::OSCData(int i){
      | ^~~~~~~
/Users/rvendrame/Documents/Arduino/libraries/OSC/OSCData.cpp:34:1: note: 'OSCData::OSCData(intOSC_t)' previously defined here
   34 | OSCData::OSCData(intOSC_t i){
      | ^~~~~~~
In file included from /Users/rvendrame/Documents/Arduino/libraries/OSC/OSCMessage.h:29,
                 from /Users/rvendrame/Documents/Arduino/libraries/OSC/OSCBundle.h:29,
                 from /Users/rvendrame/Documents/Arduino/libraries/OSC/OSCBundle.cpp:26:
/Users/rvendrame/Documents/Arduino/libraries/OSC/OSCData.h:124:5: error: 'OSCData::OSCData(int)' cannot be overloaded with 'OSCData::OSCData(intOSC_t)'
  124 |     OSCData (int);
      |     ^~~~~~~
/Users/rvendrame/Documents/Arduino/libraries/OSC/OSCData.h:121:3: note: previous declaration 'OSCData::OSCData(intOSC_t)'
  121 |   OSCData (intOSC_t);
      |   ^~~~~~~
In file included from /Users/rvendrame/Documents/Arduino/libraries/OSC/OSCMessage.h:29,
                 from /Users/rvendrame/Documents/Arduino/libraries/OSC/OSCMessage.cpp:26:
/Users/rvendrame/Documents/Arduino/libraries/OSC/OSCData.h:124:5: error: 'OSCData::OSCData(int)' cannot be overloaded with 'OSCData::OSCData(intOSC_t)'
  124 |     OSCData (int);
      |     ^~~~~~~
/Users/rvendrame/Documents/Arduino/libraries/OSC/OSCData.h:121:3: note: previous declaration 'OSCData::OSCData(intOSC_t)'
  121 |   OSCData (intOSC_t);
      |   ^~~~~~~
exit status 1

Compilation error: exit status 1
```