CNMAT/OSC #156: Intermittent INVALID_OSC Error When Parsing Identical OSC Messages on RP2040w with Arduino-Pico Core

DavidGoedicke, 2024-11-30:
Hello,

I'm working on a project using an RP2040w programmed with Arduino, utilizing the [arduino-pico](https://github.com/earlephilhower/arduino-pico) core by Earle Philhower. I'm encountering an issue with decoding OSC messages on one of the cores.

**Problem**:

I'm trying to decode simple OSC messages, but the same message—consisting of identical bytes—sometimes decodes successfully and sometimes results in an error. Specifically, I receive error code 2, which corresponds to INVALID_OSC in the OSC library.

Below is an example where I sent the same message twice. The first attempt fails, and the second decodes successfully.

Is there something I might be missing in how I'm handling the OSC messages on the RP2040w? Could this be a bug in the OSC library, or perhaps an issue with the way UDP packets are being read and parsed?

Any pointers or suggestions would be greatly appreciated!

Thank you so much for your time and assistance!

Example output
```
WiFi connected
IP address: 
192.168.2.96
size12
0:47 ('/')
1:105 ('i')
2:110 ('n')
3:0 ('')
4:44 (',')
5:105 ('i')
6:0 ('')
7:0 ('')
8:0 ('')
9:0 ('')
10:0 ('')
11:66 ('B')
The message had an error: 2 (INVALID_OSC)

size12
0:47 ('/')
1:105 ('i')
2:110 ('n')
3:0 ('')
4:44 (',')
5:105 ('i')
6:0 ('')
7:0 ('')
8:0 ('')
9:0 ('')
10:0 ('')
11:66 ('B')
Got an OSC Message
Received value: 66
```

The receiving code:
```cpp
#ifdef USEOSC

  int size = Udp.parsePacket();
  if (size > 0) {
    OSCBundle inOSCMsg;
    inOSCMsg.empty();
    delay(1);
    Serial.print("size");
    Serial.println(size);
    uint8_t buffer[64];  // Adjust the size according to your needs
    if (size <= sizeof(buffer)) {
      Udp.read(buffer, size);

      // Debug: print the buffer content
      for (int i = 0; i < size; i++) {
        int b = buffer[i];
        Serial.print(i);
        Serial.print(":");
        Serial.print(b);
        Serial.print(" ('");
        Serial.print((char)b);
        Serial.println("')");
      }

      inOSCMsg.fill(buffer, size);
      

      if (!inOSCMsg.hasError()) {
        Serial.println("Got an OSC Message");
        // Process the message
        oscCallback(inOSCMsg.getOSCMessage(0));
      } else {
        Serial.print("The message had an error: ");
        Serial.print(inOSCMsg.getError());
      }
      inOSCMsg.empty();
    } else {
      Serial.println("Error: Packet too large for buffer.");
    }
  }
#endif

adrianfreed, 2024-11-30:
I wonder if it is because you are receiving a simple message but it expects a bundle.

adrianfreed, 2024-12-02:
I merged a change which improves handling of messages which my help

DavidGoedicke, 2024-12-19:
Thank and actually I didn't realize there was a difference. I could fix it. Thank you for your help!!

DavidGoedicke, 2024-12-25:
I think we can close this issue.