CNMAT/OSC #159: Raspberry Pi Pico Serial Communication

jamiejcole, 2024-12-31:
I cannot get my RP2040 Pico to successfully handshake with ETC EOS using OSC over Serial. I have used this library before on an Arduino UNO and an ESP8266 working perfectly. 

I tried uploading the following test sketch to my Pico using the Arduino IDE and Earle Philhower's [Arduino Pico SDK](https://github.com/earlephilhower/arduino-pico). This sketch works **if and only if the Serial Monitor is opened and closed** prior to launching EOS on Windows 11 24H2, however works as expected on MacOS. I found a post on the #lighthack forums identifying a [similar issue](https://community.etcconnect.com/etclabs/f/lighthack/37839/lighthack-with-raspberry-pico).

```c
#include <OSCBoards.h>
#include <OSCBundle.h>
#include <OSCData.h>
#include <OSCMatch.h>
#include <OSCMessage.h>
#include <OSCTiming.h>
#ifdef BOARD_HAS_USB_SERIAL
  #include <SLIPEncodedUSBSerial.h>
    SLIPEncodedUSBSerial SLIPSerial(thisBoardsSerialUSB);
  #else
  #include <SLIPEncodedSerial.h>
    SLIPEncodedSerial SLIPSerial(Serial);
#endif
#include <string.h>

const String HANDSHAKE_QUERY = "ETCOSC?";
const String HANDSHAKE_REPLY = "OK";

void parseOSCMessage(String& msg)
{
  if (msg.indexOf(HANDSHAKE_QUERY) != -1)
  {
    SLIPSerial.beginPacket();
    SLIPSerial.write((const uint8_t*)HANDSHAKE_REPLY.c_str(), (size_t)HANDSHAKE_REPLY.length());
    SLIPSerial.endPacket();
  }
}

void setup()
{
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
  SLIPSerial.begin(115200);
  while (!Serial)
    continue;
  
  delay(500);
  SLIPSerial.beginPacket();
  SLIPSerial.write((const uint8_t*)HANDSHAKE_REPLY.c_str(), (size_t)HANDSHAKE_REPLY.length());
  SLIPSerial.endPacket();
}

void loop()
{
  static String curMsg;
  static unsigned long lastTimeSent;
  static int32_t pingNum;
  unsigned long curTime;
  int size;

  // Check to see if any OSC commands have come from Eos that we need to respond to.
  size = SLIPSerial.available();
  if (size > 0)
  {
    // Fill the msg with all of the available bytes
    while (size--)
      curMsg += (char)(SLIPSerial.read());
  }
  if (SLIPSerial.endofPacket())
  {
    parseOSCMessage(curMsg);
    curMsg = String();
  }

  // Send a ping every second.
  curTime = millis();
  if (curTime - lastTimeSent > 1000)
  {
    OSCMessage ping("/eos/ping");
    ping.add((intOSC_t)(pingNum++));
    SLIPSerial.beginPacket();
    ping.send(SLIPSerial);
    SLIPSerial.endPacket();
    lastTimeSent = curTime;
  }
}
```

With the Arduino IDE closed and the Pico unplugged from my machine, if I open EOS and then plug the Pico in, EOS recognises the device and attempts to send the `ETCOSC?` message, but receives no `OK` reply from the Pico. The `OK` message is only sent if the Serial Monitor has been opened and closed prior to launching EOS.

After spending some time using Wireshark and usbpcap, I have discovered the following:
1. On an Arduino UNO, the serial port is recognised as using the `USB` protocol, receives the `ETCOSC?` handshake request, and replies accordingly with `OK`:
[image]

2. Using the Pico, the serial port uses the `USBCOM` protocol instead. The Pico correctly receives the `ETCOSC?` request (highlighted in screenshot), but doesn't reply at all.
[image]

I have tried using MicroPython and CircuitPython on the Pico to communicate with EOS to no avail. Any input would be much appreciated. Any thoughts @sstaub?

sstaub, 2024-12-31:
Try to comment out following code on Pico
ˋˋˋ
  while (!Serial)
    continue;
  
  delay(500);
  SLIPSerial.beginPacket();
  SLIPSerial.write((const uint8_t*)HANDSHAKE_REPLY.c_str(), (size_t)HANDSHAKE_REPLY.length());
  SLIPSerial.endPacket();

`ˋˋ

jamiejcole, 2024-12-31:
[image]
[image]

Thanks for the quick reply. Still no luck unfortunately. I tried turning an LED on when the `ETCOSC?` message is received, and I can confirm that it does turn on, so the Pico definitely receives messages, just can't send them.
```c
void parseOSCMessage(String& msg)
{
  if (msg.indexOf(HANDSHAKE_QUERY) != -1)
  {
    digitalWrite(LED_BUILTIN, HIGH);
    SLIPSerial.beginPacket();
    SLIPSerial.write((const uint8_t*)HANDSHAKE_REPLY.c_str(), (size_t)HANDSHAKE_REPLY.length());
    SLIPSerial.endPacket();
  }
}
```

sstaub, 2024-12-31:
A will have look into the next days, because I use normal macOS and don't test my library https://github.com/sstaub/eOS on windows, but I will try. I don't think this a problem of the OSC library, it is more the start code.

jamiejcole, 2024-12-31:
Thanks, I appreciate it. I'll have a try with your eOS library and see if I can get it working.

sstaub, 2024-12-31:
I think the same problem exist also on my library. Because I write a new one I will test it also on Windows in future.
Can you place here your .ino file of your code.

jamiejcole, 2024-12-31:
Sure no worries, here is [my ino file](https://gist.github.com/jamiejcole/113ef2697e0bc6ee0a285b755a203782). I don't think it's an issue with Wireshark as I have tried it on a separate machine as well. Let me know if I can help at all

sstaub, 2024-12-31:
I tried your file on Windows11 with Arduino IDE 2.3.4 using UF2 upload method. And it works.
Maybe this is a problem using the Pico uploading from different OS.
It is not a problem of your code, this problem comes from the Arduino Pico Core.

jamiejcole, 2025-01-01:
I've also just tried it on another Windows laptop and it works too - what??? I tried uninstall Wireshark/usbpcap and other USB emulators that could be interfering with it on my main machine but nothing. So confused... Anyway, thank you very much for the help, I would've been pulling my hair out for the next week otherwise.

closed 2025-01-01 by jamiejcole